kenzok8/openwrt_Build #9: How to output openwrt-x86-64-generic-ext4-combined.img.gz file?

drejohnson, 2021-11-11:


kenzok8, 2021-11-12:
![Uploading photo_2021-11-12_14-49-59.jpg…](
[image]
)

kenzok8, 2021-11-12:
![photo_2021-11-12_14-49-59](https://user-images.githubusercontent.com/39034242/141423117-d686a572-99a0-446d-b108-bd87d47335a2.jpg
[image]
)

kenzok8, 2021-11-12:
[image]

kenzok8, 2021-11-12:
[image]

closed 2021-11-23 by kenzok8