kenzok8/openwrt_Build #18: About date format and filename

fengertao, 2022-04-10:
1. 日期上是否可以添加年
比如编译部署好以后，主页面显示
固件版本：              Openwrt v04.09 / kenzo (April)
时间长了我就不记得是2020年4月9日，还是2021年，还是2022年做的固件。

2. 文件名是否可以把openwrt换成lean/lienol等
比如
https://op.dllkids.xyz/op/firmware/Lean/03.25-openwrt-x86-64-generic-squashfs-combined-efi.img.gz
是否可以改名为2022.03.25-lean-x86-64-generic-squashfs-combined-efi.img.gz
这样我下载以后，就知道是哪年下载的，是lean大的还是lienol大的固件。

kenzok8, 2022-04-14:
1.date格式更改下就行默认date +%m.%d   加个%Y就可以了，看日期其实最简单的办法你可以在内核日志里看，可以看到编译时间
2、更名的话也简单，脚本写入下就好了
但是这两个有意义吗？

fengertao, 2022-04-15:
[image]
如图,这个正在使用的软路由,是2022年4月9日的build,还是2021年4月9日?还是2020年4月9日?过几个月就搞不清了.

这个固件到底是Lienol大的,还是Lean大的?也不知道了.

closed 2022-04-15 by kenzok8