kenzok8/openwrt_Build #21: 请问如何设置动态链路聚合

qyingkou, 2022-04-27:
请教ken大一个openwrt的设置问题。

当前背景：
我的交换机是Tplink sg-3218,支持双网口的动态链路聚合
我的软路由是NanoPi R4S，安装Openwrt v01.18 / kenzo (Mar)固件，作为旁路由使用（仅用了一个网口）

我的问题：
我想把旁路由（r4s）的两个网口都用起来，通过动态链路聚合实现双网口叠加速度。
当前对openwrt设置实在不熟悉，摸索了一阵子还是没有思路。
我想请问：
- 该版本固件，支持动态链路聚合设置吗
- openwrt的网络接口该如何设置，才能实现动态链路聚合

感谢老大指路

qyingkou, 2022-04-28:
今天又在该系统下找了一遍还是没找到网口聚合选项，期待ken大抽空给结论。🙏

kenzok8, 2022-05-01:
我不知道你的需求是wan口的汇聚还是lan口汇聚
另外建议你网上找找教程吧，
https://post.smzdm.com/p/av7o06mn/

qyingkou, 2022-05-01:
因为作为旁路由使用，只用到了lan，wan是空着的。
我希望将wan口当作另外一个lan2，把lan1和lan2动态链路到交换机实现2Gbps的带宽。

想问问您的资源站中可否添加以下 luci-proto-bonding 这个协议插件（支持动态链路聚合）呢？谢谢啦

masx200, 2023-03-22:
> 谢谢提供的教程，受用。 我希望将这两个口同时作为lan口来使用（即双lan口聚合，连接2条线到交换机），实现2Gbps的带宽。
> 
> 您的资源站中有文章中所说的这三个协议插件吗：
> 
> * kmod-bonding
> * proto-bonding
> * luci-proto-bonding
> 
> 系统软件包中可以找到他们但无法安装成功，继续求助

听说内核模块要重新编译进去才行

masx200, 2023-03-22:
https://www.right.com.cn/forum/thread-534350-1-1.html

--force-depends

closed 2023-04-29 by kenzok8