kenzok8/openwrt_Build #33: AdGuardHome 初始化打不开配置网页

geopenwu, 2023-12-21:
安装好openwrt，启动AdGuardHome后，点开AdGuardHome:3000, 会出现404错误。如附件：

### 启动AdGuardHome：
[image]

### 错误页面：
[image]

liawt, 2023-12-26:
1. 看日志处是否提示端口被占用, 如果被占用则在配置页面把端口改成类似于3001
2. 改完之后新端口还是不行请点击更新核心

liawt, 2023-12-26:
我也刚下Lean版的固件.发现第一次网页打开也是提示30000被占用, 后来更改端口以及升级核心后解决.
但是恢复快照重试之后发现升级核心功能总是失败, 最后通过软件包里面的Adgurad升级到最新版之后发现直接用30000端口也能打开了. 很奇怪.

推测日志提示的30000端口占用实际上本身就是Adgurad占用的, 但是默认编译的版本有点问题. 所以通过Adgurad自带升级功能或者是软件包升级覆盖安装之后原来的30000端口也能用了

kenzok8, 2026-02-24:
自动巡检：该 Issue 长时间未更新（>=180 天），先关闭以便集中处理当前问题。

如问题仍存在，请按最新源码重新复现后提交新 Issue，并附：
- 固件版本与目标架构
- 完整编译/运行日志（从第一个报错开始）
- 复现步骤

Auto triage: this issue has been inactive for >=180 days, so it is being closed for backlog cleanup.
If the issue still exists, please open a new issue with updated logs and clear reproduction steps.